nxp-mcuxpresso/legacy-mcux-sdk #193: add control of PixelPattern for imxrt pxp driver

glory-man, 2024-05-03:
Is it possible to add `ELCDIF_SetPixelPattern()` function to `drivers/pxp/fsl_pxp.c`
```
/*!
 * @brief eLCDIF configure pixel pattern.
 */
typedef enum _elcdif_pixel_pattern_config
{
    kELCDIF_PixelPatternRGB = 0x0,  /*!<  0b000..RGB. */
    kELCDIF_PixelPatternRBG = 0x1,  /*!<  0b001..RBG. */
    kELCDIF_PixelPatternGBR = 0x2,  /*!<  0b010..GBR. */
    kELCDIF_PixelPatternGRB = 0x3,  /*!<  0b011..GRB. */
    kELCDIF_PixelPatternBRG = 0x4,  /*!<  0b100..BRG. */
    kELCDIF_PixelPatternBGR = 0x5,  /*!<  0b101..BGR. */
    kELCDIF_PixelPatternLast,       /*!<  unavalable configuration. */
} elcdif_pixel_pattern_config_t;

/*!
 *  brief Set the order of the RGB components of each pixel in lines.
 *
 * param base eLCDIF peripheral base address.
 * param pixelPattern The pixel pattern
 */
void ELCDIF_SetPixelPattern(LCDIF_Type *base, elcdif_pixel_pattern_config_t pixelPattern)
{
    assert((uint32_t)pixelPattern < kELCDIF_PixelPatternLast);
    
    base->CTRL2_CLR = (LCDIF_CTRL2_CLR_EVEN_LINE_PATTERN_MASK | LCDIF_CTRL2_CLR_ODD_LINE_PATTERN_MASK);
    base->CTRL2_SET = ( (pixelPattern << LCDIF_CTRL2_SET_ODD_LINE_PATTERN_SHIFT) | (pixelPattern << LCDIF_CTRL2_SET_EVEN_LINE_PATTERN_SHIFT) );
}
```
Current version of driver doesn't have such function while imxrt pxp module can control pixel pattern.

mcuxsusan, 2024-05-06:
Thanks for the suggestion of improvement, already forwarded to development team for check. Appreciate your patience since reply could be delayed.

glory-man, 2025-06-23:
is any info about this ?