nxp-mcuxpresso/legacy-mcux-sdk #194: latest SDK release 2.15.100 missing here

raveslave, 2024-05-05:
hi, noted that latest SDK release 2.15.100 missing here while it's released on https://mcuxpresso.nxp.com since  26 Apr 2024

github should be first in my opinion?

mcuxsusan, 2024-05-06:
Hello @raveslave, we have tagged the https://github.com/nxp-mcuxpresso/mcux-sdk/releases/tag/MCUX_2.15.100 for 2.15.100 release, release has not be announced yet.
The github release happens very close to MCUXpresso SDK package release but it could be several days later.

closed 2024-05-06 by mcuxsusan

raveslave, 2024-05-06:
thanks! except for comparing with the previous release-tag (here on github), 

is there an official change-notes file for new SDK releases?   
i had a look in the pdf that comes in the mcuxpresso web-download, in that one it's quite hard to follow what is new for a specific mcu or a specific release-tag.

also curious about middleware, like usb stacks. as those are not part of this repo, where is the corresponding tags for other deps part of a SDK release?

raveslave, 2024-05-06:
replying my second question: https://github.com/nxp-mcuxpresso/mcux-sdk-middleware-usb/compare/MCUX_2.15.000...MCUX_2.15.100  
seems there are no changes in .100

mcuxsusan, 2024-05-08:
Hi @raveslave, for the usb middleware, comparing to MCUX_2.15.000, no change, the change only relates to manifest version upgrade. See https://github.com/nxp-mcuxpresso/mcux-sdk/releases/tag/MCUX_2.15.100 gives a brief summary.
Currently for Github SDK we did not have changelog provided, the one you downloaded from mcuxpresso.nxp.com has the changelog showed but that's not a diff. We are in the process of documentation improvement, hopefully we could get better documentation targeting this September.