nxp-mcuxpresso/legacy-mcux-sdk #196: Getting started page references mingw for windows but current windows does not support mingw

joeatbayes, 2024-05-08:
**Is your enhancement proposal related to a problem? Please describe.**
[Getting started page references mingw for windows but current 64 bit windows does not support mingw](https://github.com/nxp-mcuxpresso/mcux-sdk/blob/main/docs/run_a_project_using_armgcc.md)

**Describe the solution you'd like**
It would be better to convert this example to generic Linux or if you want to remain Windows compatible then port to the example to WSL2 under windows with WSL hosting Ubuntu.      

A version that works on the current version of ubuntu linux running native would also be helpful

**Describe alternatives you've considered**
In some cases I have ran into issues with programmers and other serial devices working correctly under WSL.   It may be better to use a Docker image running linux with the build tools.  That way it is easier to ensure compatibility with underlying os versions as they evolve.

mcuxsusan, 2024-05-28:
Hi @joeatbayes, thanks for your advice and sorry for the late reply. We are planning to move to ninja starting from Q3, which has more friendly installation for Windows instead of mingw. Does that make sense for you?

mcuxsusan, 2024-05-28:
by the way, do you mean windows 64-bit not support MinGW any more? We were not aware of that.