nxp-mcuxpresso/legacy-mcux-sdk #198: USB stack using freertos task and external message queue.

anischali, 2024-06-20:
We implemented a full peripheral power management system, so we disable and enable devices multiple times and we also use a the usb use task feature. doing that we might have an unexpected behavior with the freertos heap allocation since deinit usb stack will destroy the internal msg queue and create it again and again. with the other devices we separated the memory resources from the rest so we allocate memory only one at the init time but in case of usb stack the sdk is responsible for allocating and destroying the queues. 

**The solution:**

1. would be an external (from sdk) provided queue when calling the init code just like the usb use task macro.
2. A macro that envelops the create and destroy queue and when used we don't destroy the queue in runtime to keep the same queue all the time (I have already tested it and I can eventually submit a merge request if you want).

Thank you.

mcuxsusan, 2024-06-26:
Hi @anischali, thanks for raising the issue. I have forwarded the issue to develop team, the feedback could be delayed.

MarkWangChinese, 2025-09-25:
Hi @anischali I think you have the same requirement as https://github.com/nxp-mcuxpresso/mcuxsdk-middleware-usb/issues/4#issuecomment-3328176473 Please check whether the https://github.com/nxp-mcuxpresso/mcuxsdk-middleware-usb/issues/4#issuecomment-3328176473 can resolve your issue.

The concept is:
- USB device stack task function has no `while`, so the application need to add `while` to USB device stack function.
- In the `while`, you can check whether the task needs to be terminated. If yes, application can exit the `while`, then terminate self task.
- The usb device stack can be terminated by calling USB stack APIs.