nxp-mcuxpresso/legacy-mcux-sdk #199: [BUG] Missing CMSIS RTOS abstraction for i.MX RT

stefanct, 2024-06-27:
**Describe the bug**

We would like to port a library to our project that relies on the functions declared in `cmsis_os2.h`. I can see this file is included in the SDK and referenced by the manifests of the RT series including the rt1020 that we are using. However, neither within Eclipse nor within a CMake-based project it seems to be available (i.e., the include directory is not added to the compiler command). From the CMake code it looks like maybe it was a user-visible/selectable component (`set(CONFIG_USE_CMSIS_Device_API_RTOS2 true)` is contained in the RTs' `all_lib_device.cmake`). There doesn't seem to be any example in the examples repository that uses it but I see no explicit indication whatsoever that it shouldn't be available. I presume it is a bug. Do I missing something?

**To Reproduce**

- Environment (please complete the following information):
  - This is at least as described in 2.15 and 2.15.1
  - Toolchain: [e.g. ARMGCC 9-2020-q2-update]
  - Board/SoC: rt1020 evk
- Steps to reproduce the behavior:
  Add an `#include <cmsis_os2.h>` preprocessor statement and try to compile.

**Expected behavior**
A working build :)

**Additional context**
I haven't checked at all if the actual library is available for linking yet.

mcuxsusan, 2024-07-02:
Hi @stefanct,  currently the CMSIS RTOS2 is not used by any of the SDK examples.
For your case, I see you would like to include the cmsis_os2.h but it's not included as expected, am I right? I did a try to set(CONFIG_USE_CMSIS_Device_API_RTOS2 true) in the hello_world config.cmake for the rt1020 evk board and it seems OK.
```

diff --git a/evkmimxrt1020/demo_apps/hello_world/armgcc/config.cmake b/evkmimxrt1020/demo_apps/hello_world/armgcc/config.cmake
index c9c7004db5..b35e3eb9bf 100644
--- a/evkmimxrt1020/demo_apps/hello_world/armgcc/config.cmake
+++ b/evkmimxrt1020/demo_apps/hello_world/armgcc/config.cmake
@@ -28,3 +28,4 @@ set(CONFIG_KIT evkmimxrt1020)
 set(CONFIG_DEVICE_ID MIMXRT1021xxxxx)
 set(CONFIG_FPU DP_FPU)
 set(CONFIG_DSP NO_DSP)
+set(CONFIG_USE_CMSIS_Device_API_RTOS2 true)
diff --git a/evkmimxrt1020/demo_apps/hello_world/hello_world.c b/evkmimxrt1020/demo_apps/hello_world/hello_world.c
index 2daf4795ab..7814a731fd 100644
--- a/evkmimxrt1020/demo_apps/hello_world/hello_world.c
+++ b/evkmimxrt1020/demo_apps/hello_world/hello_world.c
@@ -11,6 +11,7 @@
 #include "pin_mux.h"
 #include "clock_config.h"
 #include "board.h"
+#include "cmsis_os2.h"

```

stefanct, 2024-07-03:
That's basically what I have tried in my own project. However, in my case the config.cmake gets overwritten with the CONFIG_USE_CMSIS_Device_API_RTOS2 line removed when launching kconfig/menuconfig and before/while building. You probably can already tell where this is going by now but I'll continue explaining how I came to my own conclusions:

I could re-create the build error in the hello world example you mentioned but only in an older version. Namely before `41cac9cfafa67e025d9095e2c3fa37c4bf31d85f` when `include(${SdkRootDirPath}/core/cmake/mcux.cmake)` has been removed from the example's `CMakeLists.txt`. I always viewed `mcux.cmake` as a central component of the build system and have based my own build system around it. However, apparently this was a misunderstanding on my side and the examples and actually basically everything in the SDK are independent from `mcux.cmake` and it is unused at the moment. The applications are supposed to directly include `core/devices/*/all_lib_device.cmake` and pass the `core/tools/cmake_toolchain_files/armgcc.cmake` toolchain file manually to cmake. Is that correct?

Assuming I have now grasped the situation better I wonder how to continue. You wrote in #179 your team is working on a re-design of the whole kconfig integration. Is there a timeline for this? I'd rather not migrate away from mcux.cmake at this point as I spent quite some time to make it work satisfactorily in my environment :) So I'd rather look at the kconfig implementation myself to fix this include file problem but any comments/insight from you are very welcome. Thank you.

stefanct, 2024-07-03:
Oh and I forgot to mention that we had this problem initially in Eclipse on a pristine system that does not use my build system. In Eclipse there is simply no way to add `cmsis_os2.h`/the `CMSIS_Device_API_RTOS2` component at all AFAICT. This kind of convinced me that my build system is not responsible whatsoever. I am not sure if this is off-topic here though.