nxp-mcuxpresso/legacy-mcux-sdk #200: [BUG] FSL_FEATURE_FLEXSPI_HAS_RESET never defined

JonathonReinhart, 2024-06-28:
**Describe the bug**
As far as I can tell, `FSL_FEATURE_FLEXSPI_HAS_RESET` is not defined for any devices supported by this SDK. See [this code search](https://github.com/search?q=repo%3Anxp-mcuxpresso%2Fmcux-sdk+FSL_FEATURE_FLEXSPI_HAS_RESET&type=code).

This flag is used to guard whether `FLEXSPI_Init()` will reset the flexspi block:

https://github.com/nxp-mcuxpresso/mcux-sdk/blob/1cb5377bf2bf4a7b1b6415c662100dc01b981e07/drivers/flexspi/fsl_flexspi.c#L102-L105

and here:

https://github.com/nxp-mcuxpresso/mcux-sdk/blob/1cb5377bf2bf4a7b1b6415c662100dc01b981e07/drivers/flexspi/fsl_flexspi.c#L265-L268

The result is that `FLEXSPI_Init()` does not reset the FlexSPI as it is clearly intended to do.

Since the reset is asserted by default (on RT595), one must manually reset the peripheral (or clear the reset line) before `FLEXSPI_Init()` in order to to use it, e.g.
```c
RESET_PeripheralReset(kFLEXSPI1_RST_SHIFT_RSTn);
```

Albort12138, 2024-07-02:
Hi @JonathonReinhart, thanks for your comments. That's indeed an issue, I will fix it as soon as possible.

closed 2024-07-04 by McuxCIBot

JonathonReinhart, 2024-07-04:
In b4dca30, you defined `FLEXSPI_RESETS_ARRAY` to `FLEXSPI_RSTS`. Any reason to not simply use `FLEXSPI_RSTS` instead?

Albort12138, 2024-07-05:
> In [b4dca30](https://github.com/nxp-mcuxpresso/mcux-sdk/commit/b4dca3063ec83136d9f9cb3d42b05e8b1bcdf455), you defined `FLEXSPI_RESETS_ARRAY` to `FLEXSPI_RSTS`. Any reason to not simply use `FLEXSPI_RSTS` instead?

To align with format of other drivers.

commit b4dca306 (2024-07-04): fix #200: FSL_FEATURE_FLEXSPI_HAS_RESET never defined.

- Updated code of reset peripheral.