nxp-mcuxpresso/legacy-mcux-sdk #213: [BUG] CSI2-RX does not work after having deinitialized and initialized

trunghieulenxp, 2024-10-10:
**Describe the bug**

**To Reproduce**
After having called function in fsl_mipi_csi2rx.c `void CSI2RX_Deinit(MIPI_CSI2RX_Type *base)` (https://github.com/nxp-mcuxpresso/mcux-sdk/blob/6958175a32bb12384e3635e28343c33daa3b2a42/drivers/mipi_csi2rx/fsl_mipi_csi2rx.c#L250)

and when we call again the function `void CSI2RX_Init(MIPI_CSI2RX_Type *base, const csi2rx_config_t *config) ` (https://github.com/nxp-mcuxpresso/mcux-sdk/blob/6958175a32bb12384e3635e28343c33daa3b2a42/drivers/mipi_csi2rx/fsl_mipi_csi2rx.c#L220)

The module is halted completely. We have inspected the code, and we found that by removing:
`MIPI_CSI2RX_SoftwareReset(base, true)` (https://github.com/nxp-mcuxpresso/mcux-sdk/blob/6958175a32bb12384e3635e28343c33daa3b2a42/drivers/mipi_csi2rx/fsl_mipi_csi2rx.c#L254)

We can deinit and init the module without problem.

- Environment (please complete the following information):
  - Tag/Commit hash: 6958175a32bb12384e3635e28343c33daa3b2a42
  - Toolchain:
  - Board/SoC: IMXRT 1170 EVKB
  - Example:

zejiang0jason, 2024-10-11:
Hi @trunghieulenxp , thanks for report, can you access the SDK 2.16 package? I tried with the csi_mipi_rgb project, modify the code here 
https://github.com/nxp-mcuxpresso/mcux-sdk-examples/blob/fc18ff0c12ba90fe71714eaa4244717efa18c61c/evkbmimxrt1170/driver_examples/csi/mipi_rgb/cm7/camera_support.c#L242

 to
 
```
    CSI2RX_Deinit(MIPI_CSI2RX);
    CSI2RX_Init(MIPI_CSI2RX, &csi2rxConfig);
```
but didn't reproduce.

trunghieulenxp, 2024-10-11:
Hi Jason, thanks for your answer, this error only appeared on the second time you do which means you need to call:
`CSI2RX_Init(MIPI_CSI2RX, &csi2rxConfig)
`
to start, and leave it work for at least one frame and then call:
`    CSI2RX_Deinit(MIPI_CSI2RX);
`
For stopping the process. 

After that you call again:
`CSI2RX_Init(MIPI_CSI2RX, &csi2rxConfig)
`