nxp-mcuxpresso/legacy-mcux-sdk #232: Missing macro definition in MKL25Z4_features header file

WajdiMuh, 2025-03-30:
Hello,

https://github.com/nxp-mcuxpresso/mcux-sdk/blob/a1bdae309a14ec95a4f64a96d3315a4f89c397c6/drivers/tpm/fsl_tpm.h#L37

im new to this SDK however if i understand correctly the `FSL_FEATURE_TPM_HAS_32BIT_COUNTERn` macro is defined the features header file of different MCUs such as in https://github.com/nxp-mcuxpresso/mcux-sdk/blob/a1bdae309a14ec95a4f64a96d3315a4f89c397c6/devices/MKW41Z4/MKW41Z4_features.h#L1840

however since this macro is not defined in https://github.com/nxp-mcuxpresso/mcux-sdk/blob/main/devices/MKL25Z4/MKL25Z4_features.h

that means i cant use the TPM driver for this MCU. i believe a messing definition of this macro in the MKL25Z4 header file is needed.

Best Regards
Wajdi

WajdiMuh, 2025-03-30:
I trying to add support for the TPM peripheral for the MKL25Z4 in Zephyr. as there was no definition of this peripheral for some reason even though the MCU has support for this peripheral.

mcuxsusan, 2025-03-31:
Hi @WajdiMuh, 

We plan to deprecate this repository around June. Please move to our new SDK project [mcuxsdk](https://github.com/nxp-mcuxpresso/mcuxsdk-manifests) for any future contributions, as we will stop updating this one.

I noticed that this issue is related to the legacy SoC MKL25Z4 support in Zephyr. Please report this issue to [hal_nxp](https://github.com/zephyrproject-rtos/hal_nxp) since we no longer provide support for legacy SoCs in our new SDK. Sorry for the inconvenience.

closed 2025-03-31 by mcuxsusan