nxp-mcuxpresso/legacy-mcux-sdk #235: fsl_adapter_lpuart compilation error

Wielebny666, 2025-04-15:
https://github.com/nxp-mcuxpresso/mcux-sdk/blob/0420001d787c2c7bb062a2da4c46233d9daf2737/components/uart/fsl_adapter_lpuart.c#L1924

'edma_tcd_t' {aka 'struct _edma_core_tcd'} has no member named 'CSR'GCC
struct "_edma_core_tcd" has no field "CSR"

zejiang0jason, 2025-04-16:
Hi @Wielebny666 , which board are you using? We can check it.

The SDK repos have moved to https://github.com/nxp-mcuxpresso/mcuxsdk-manifests, the patches will be updated in the new repos.

Thanks.

Wielebny666, 2025-04-16:
MCXA153

mcuxsusan, 2025-04-16:
Hello @Wielebny666,

We plan to deprecate this repository around June. Please move to our new SDK project [mcuxsdk](https://github.com/nxp-mcuxpresso/mcuxsdk-manifests) for any issue, as we will stop updating this one.